intel/ros_intel_movidius_ncs #48: [launch files] realsense_camera pkg OR realsense_ros_camera pkg

jim1993, 2017-12-28:
Hi
The "ncs_camera.launch" file in "movidius_ncs_launch" is trying to load realsense driver from "realsense_ros_camera pkg" (line 72), however, the latest ros package for realsense driver now is "realsense camera pkg". 
I have successfully executed the video streaming demo with SR300 by modifying the launch file (replaced "realsense_ros_camera" with  "realsense camera" in line 72). So, maybe you guys can fix this little bug.

Cheers

dongx1x, 2018-01-02:
@jim1993 
The project name of realsense [v2.0.0](https://github.com/intel-ros/realsense/tree/2.0.0) or the latest version [v2.0.1](https://github.com/intel-ros/realsense/tree/2.0.1) has changed to "realsense_ros_camera".

xhuan28, 2018-01-31:
Currently, our development is focusing on RealSense D400 serials which is using librealsense2, so we will keep it unchanged.

closed 2018-01-31 by xhuan28