intel/ros_intel_movidius_ncs #56: Python3 vr Python2.7 ROS Conflict.

sradmard, 2018-01-31:
Hi,
I am wondering whether this ROS package can work in case only the movidius api is installed instead of its entire SDK. The reason is that it is problematic to install its SDK on an embedded board, while the api can be easily installed and set. 

My second challenge is that I have developed a simple ROS wrapper using python for the movidius when I have the api installed only. However, I can only get some models like mobilenet to work due to the python version conflict. ROS only supports python2.7, and some movidius models like Inception only support python3. Do you have any idea how to find a workaround for this? 
Thank you so much in advance.

xhuan28, 2018-01-31:
@sradmard 
1. You can disable toolkit installation by setting "no" to "INSTALL_TOOLKIT" in ncsdk.conf, so that Caffe and Tensorflow won't be installed as well.
2. NC API already supports Python2.7 from version1.09, you can refer the its readme file for details.https://github.com/movidius/ncsdk/blob/v1.09.00.06/docs/release_notes.md
Although the example models are developed under python3, you can adjust the scripts to run under python2.7.

sradmard, 2018-02-01:
Thank you so much for your response. However, I still have the following confusion:
1. Would disabling "INSTALL_TOOLKIT" in the ncsdk installation be the same as the api installation described here:
[https://movidius.github.io/blog/ncs-apps-on-rpi/](https://movidius.github.io/blog/ncs-apps-on-rpi/)
Because I can easily install the api and it is quite quick on RPI, while the sdk installation takes forever. On the other hand, have you ever tried your ROS wrapper with the api mode installation only? I am not sure whether it would work or not.
2. The problem is that even though all apis are supposed to work with python2.7, not all of them return sensible results. For example, inception only seems to return correct results when run with python3. Please refer to this issue regarding that:
[https://ncsforum.movidius.com/discussion/556/tensorflow-inception-network-generates-wrong-output#latest](https://ncsforum.movidius.com/discussion/556/tensorflow-inception-network-generates-wrong-output#latest)

xhuan28, 2018-02-05:
@sradmard 
1. The method mentioned in https://movidius.github.io/blog/ncs-apps-on-rpi/ is compiling directly in the folder of api,  but it approaches the same target that not to build the toolkit in. I have tried ROS wrapper with the api mode installation both with my method and theirs. Both of them works.
2. https://ncsforum.movidius.com/discussion/556/tensorflow-inception-network-generates-wrong-output#latest mentioned that you should change the "1/128" to "1.0/128.0", it does work after changing the code. The result under python2 and python3 is same. You can try it on your side.

closed 2018-02-12 by xhuan28