intel/ros_intel_movidius_ncs #57: No device found?

j5207, 2018-02-03:
Hi all:
When I'm trying to launch any of demo on video, I found out that there is "NO devices error"
[image]
I try to switch the USB port to USB2 and it doesn't help.
I'm using Realsense D415, ubuntu 16.04, Ros Kinetic, it would be much helpful if anyone could help me out.

Thanks,

xhuan28, 2018-02-05:
@j5207 
Have you ever used your NCS successfully? This issue emerges when running mvNCCompile not executing demo. May I know which command you used?
I once encountered a device timeout issue which is caused by power supply(https://ncsforum.movidius.com/discussion/comment/1493/#Comment_1493), so I suggest you use a powered USB hub and try again.

j5207, 2018-02-05:
@xhuan28 
Thanks for your response! Well for now...., I think the reason is that I plug the camera instead of NCS

closed 2018-02-06 by xhuan28