intel/ros_intel_movidius_ncs #60: [movidius_ncs_image-1] process has died

borongyuan, 2018-02-19:
Hi,
I'm trying to run the Demo, but the node died with following output.

process[movidius_ncs_image-1]: started with pid [27435]
[ INFO] [1519004819.475200657]: use device_index = 0
[ INFO] [1519004819.477400479]: use log_level = 2
[ INFO] [1519004819.485095608]: use cnn_type_ = googlenet
[ INFO] [1519004819.488147000]: use graph_file_path = /opt/movidius/ncappzoo/caffe/GoogLeNet/graph
[ INFO] [1519004819.490475381]: use category_file_path = /opt/movidius/ncappzoo/data/ilsvrc12/imagenet1000.txt
[ INFO] [1519004819.492334990]: use network_dimension = 224
[ INFO] [1519004819.494270084]: use channel1_mean = 104.007
[ INFO] [1519004819.496297963]: use channel2_mean = 116.669
[ INFO] [1519004819.498221808]: use channel3_mean = 122.679
[ INFO] [1519004819.500211338]: use top_n = 3
[ INFO] [1519004819.502110483]: use scale = 1
[movidius_ncs_image-1] process has died [pid 27435, exit code -4, cmd /home/borong/catkin_ws/install/lib/movidius_ncs_i
mage/movidius_ncs_image __name:=movidius_ncs_image __log:=/home/borong/.ros/log/bfedfe98-1516-11e8-9e20-b88687fb2f87/mo
vidius_ncs_image-1.log].
log file: /home/borong/.ros/log/bfedfe98-1516-11e8-9e20-b88687fb2f87/movidius_ncs_image-1*.log

The log file does not exist. I've tried SDK v1.11.00 and v1.12.00, same problem. Where can I find more info?

xhuan28, 2018-02-23:
May I know what the command which you executed is? Did you use the absolute path of the image file?

borongyuan, 2018-02-23:
`roscore`
then
`roslaunch movidius_ncs_launch ncs_image.launch cnn_type:=googlenet`
and died...

I think this step is only starting the service, the image is not provided yet.

xhuan28, 2018-02-26:
You don't need to run "roscore" separately, as roslaunch does it for you. But I don't think this is the root cause of the issue you met.
I assume you used the latest commit of master branch, the following is the screen output on my side:
process[movidius_ncs_image-1]: started with pid [15443]
[ INFO] [1519608312.075294069]: use device_index = 0
[ INFO] [1519608312.075788906]: use log_level = 0
[ INFO] [1519608312.076215842]: use cnn_type_ = googlenet
[ INFO] [1519608312.076627832]: use graph_file_path = /opt/movidius/ncappzoo/caffe/GoogLeNet/graph
[ INFO] [1519608312.077032002]: use category_file_path = /opt/movidius/ncappzoo/data/ilsvrc12/imagenet1000.txt
[ INFO] [1519608312.077435307]: use network_dimension = 224
[ INFO] [1519608312.077860920]: use channel1_mean = 104.007
[ INFO] [1519608312.078290527]: use channel2_mean = 116.669
[ INFO] [1519608312.078709479]: use channel3_mean = 122.679
[ INFO] [1519608312.079093553]: use top_n = 3
[ INFO] [1519608312.079467990]: use scale = 1
[ INFO] [1519608312.079576754]: open device #0 name=2.5

It seems the problem happens when opening the device. Could you double check the connection of your NCS device to make sure it works?
Thanks.

borongyuan, 2018-02-26:
I agree. But the connection should be OK. I can run NCSDK and ncappzoo examples without problem.
I'll try to find why it cannot be opened from ROS.
Thanks

closed 2018-03-02 by xhuan28

manhha1402, 2018-03-13:
I have the same problem as the error shows here:
[movidius_ncs_nodelet/movidius_manager-1] process has died [pid 6560, exit code -4, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=movidius_manager __log:=/home/jcl-mh/.ros/log/250dd9e0-26b2-11e8-821b-901b0e0c067c/movidius_ncs_nodelet-movidius_manager-1.log].
log file: /home/jcl-mh/.ros/log/250dd9e0-26b2-11e8-821b-901b0e0c067c/movidius_ncs_nodelet-movidius_manager-1*.log
[ INFO] [1520942280.251314252]: Bond broken, exiting

tzven0, 2018-03-21:
Hey, i also have the same Problem. I can also run examples without any problems.
Calling

roslaunch movidius_ncs_launch ncs_image.launch cnn_type:=mobilenetssd

i get

process[movidius_ncs_image-2]: started with pid [29159]
[ INFO] [1521636810.858471978]: use device_index = 0
[ INFO] [1521636810.860236001]: use log_level = 0
[ INFO] [1521636810.862037600]: use cnn_type_ = mobilenetssd
[ INFO] [1521636810.863472471]: use graph_file_path = /opt/movidius/ncappzoo/caffe/SSD_MobileNet/graph
[ INFO] [1521636810.865211264]: use category_file_path = /opt/movidius/ncappzoo/data/ilsvrc12/voc21.txt
[ INFO] [1521636810.866885513]: use network_dimension = 300
[ INFO] [1521636810.868162408]: use channel1_mean = 127.5
[ INFO] [1521636810.869520429]: use channel2_mean = 127.5
[ INFO] [1521636810.871055019]: use channel3_mean = 127.5
[ INFO] [1521636810.872097811]: use top_n = 3
[ INFO] [1521636810.872992577]: use scale = 0.007843
[ INFO] [1521636810.873251271]: open device #0 name=1.5

When i then call 

rosrun movidius_ncs_example movidius_ncs_example_image_detection /opt/movidius/ncappzoo/data/images/cat.jpg

in another Terminal i get 

[ERROR] [1521636823.881998561]: failed to call service DetectObject

and in the other Terminal i get

[movidius_ncs_image-2] process has died [pid 29159, exit code -4, cmd /home/tzven0/argo/devel/lib/movidius_ncs_image/movidius_ncs_image __name:=movidius_ncs_image __log:=/home/tzven0/.ros/log/d9b9ec76-2d06-11e8-b5c4-78929c759894/movidius_ncs_image-2.log].

ChrisHughes, 2018-04-06:
I'm getting this due to an Illegal Instruction for my current CPU.
```
Thread 1 "movidius_ncs_im" received signal SIGILL, Illegal instruction.
0x00007ffff7bc9bbb in movidius_ncs_lib::NCSManager::NCSManager(int const&, int const&, movidius_ncs_lib::Device::LogLevel const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int const&, std::vector<float, std::allocator<float> > const&, float const&, int const&) ()
   from /home/ubuntu/catkin_ws/devel/lib/libmovidius_ncs_lib.so
```

My CPU info
```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 76
model name	: Intel(R) Atom(TM) x7-Z8750  CPU @ 1.60GHz
stepping	: 4
microcode	: 0x40b
cpu MHz		: 1096.125
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 3199.91
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 76
model name	: Intel(R) Atom(TM) x7-Z8750  CPU @ 1.60GHz
stepping	: 4
microcode	: 0x40b
cpu MHz		: 829.750
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 3199.91
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 76
model name	: Intel(R) Atom(TM) x7-Z8750  CPU @ 1.60GHz
stepping	: 4
microcode	: 0x40b
cpu MHz		: 740.375
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 2
cpu cores	: 4
apicid		: 4
initial apicid	: 4
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 3199.91
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 6
model		: 76
model name	: Intel(R) Atom(TM) x7-Z8750  CPU @ 1.60GHz
stepping	: 4
microcode	: 0x40b
cpu MHz		: 1079.437
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 3
cpu cores	: 4
apicid		: 6
initial apicid	: 6
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		:
bogomips	: 3199.91
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:
```