intel/ros_intel_movidius_ncs #61: remote RealSense camera

js333031, 2018-03-03:
I'd like to have D415 on a separate Ubuntu host, streaming it's data to host with NCS software. Both hosts running ROS Kinetic. What's the best way to accomplish this?

chaoli2, 2018-03-09:
@js333031 
In order accomplish your requirement, firstly you have to setup ROS environment on two machines following this [guide](http://wiki.ros.org/ROS/Tutorials/MultipleMachines). 
After that, following this [guide ](https://github.com/intel/ros_intel_movidius_ncs/blob/master/doc/video_classification.md#2-run-with-other-ros-supported-cameras-1)to launch your camera node(on the host which connects camera)  and NCS streaming node(on the host which run NCS software). Make sure the input topic is matched with the topic URL of your RGB camera. 
Then the camera node and NCS streaming node will communicate with each other via ROS topic.

chaoli2, 2018-07-05:
Close this thread since no response. Feel free to re-open this issue when need further assist.

closed 2018-07-05 by chaoli2