intel/ros_intel_movidius_ncs #62: load model

daming5432, 2018-03-05:
How can I load several models simultaneously?

daming5432, 2018-03-05:
When I load two models, the first is ok bu the second return MVNC_BUSY.

daming5432, 2018-03-05:
When I load two models, the first is ok but the second return MVNC_BUSY.

xhuan28, 2018-03-07:
Do you mean you would like to run 2 models on one NCS device? It is not supported currently. But we are developing a Multiple NCS feature that supports inference on several NCS devices with different models simultaneously. This feature will be implemented in next release which is going to release this month. Stay tuned.

xhuan28, 2018-03-09:
May I know what the user scenario is? Why you need load two CNN models on one device simultaneously.

chaoli2, 2018-07-05:
Close this thread since no response. Feel free to re-open this issue when need further assist.

closed 2018-07-05 by chaoli2