intel/ros_intel_movidius_ncs #64: Couldn't run movidius node with D435 camera, librealsense2

manhha1402, 2018-03-15:
Hi,
I tried to follow the instruction(https://github.com/intel/ros_intel_movidius_ncs/blob/master/doc/video_classification.md#googlenet) to stream D435 camera with the command:
# Option 1: RealSense
roslaunch movidius_ncs_launch ncs_camera.launch cnn_type:=alexnet camera:=realsense

and got the error:

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 748, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 720, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 654, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 684, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 588, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)

I am able to launch the realsense2 node on my computer. 
[image]

Can anyone tell me how to solve it ?
Thanks

xhuan28, 2018-03-16:
It seems some ROS packages are missing in your environment. You can try to print out more information to debug.
Here are the lines I added to File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path:

    def get_path(self, name):
        """
        :param name: package name, ``str``
        :returns: filesystem path of package
        :raises: :exc:`ResourceNotFound`
        """
        self._update_location_cache()
        #Debug
        print "+++++resource name: %s" %name
        #Debug
        if name not in self._location_cache:
            raise ResourceNotFound(name, ros_paths=self._ros_paths)
        else:
            #Debug
            print "-----resource location: %s" %self._location_cache[name]
            #Debug
            return self._location_cache[name]

And the correct output should be:
+++++resource name: movidius_ncs_launch
-----resource location: /home/intel/workspace/catkin_ws/install/share/movidius_ncs_launch
+++++resource name: rosgraph
-----resource location: /opt/ros/kinetic/share/rosgraph
... logging to /home/intel/.ros/log/5bf84300-28c4-11e8-9d29-001fc69cba12/roslaunch-intel-desktop-14719.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

+++++resource name: movidius_ncs_launch
-----resource location: /home/intel/workspace/catkin_ws/install/share/movidius_ncs_launch
+++++resource name: realsense_ros_camera
-----resource location: /home/intel/workspace/catkin_ws/install/share/realsense_ros_camera
+++++resource name: movidius_ncs_launch
-----resource location: /home/intel/workspace/catkin_ws/install/share/movidius_ncs_launch
+++++resource name: movidius_ncs_launch
-----resource location: /home/intel/workspace/catkin_ws/install/share/movidius_ncs_launch
started roslaunch server http://intel-desktop:43287/

SUMMARY
========

PARAMETERS
 * /movidius_ncs_nodelet/movidius_ncs_stream/category_file_path: /opt/movidius/nca...
 * /movidius_ncs_nodelet/movidius_ncs_stream/channel1_mean: 104.006987932
 * /movidius_ncs_nodelet/movidius_ncs_stream/channel2_mean: 116.668767617
 * /movidius_ncs_nodelet/movidius_ncs_stream/channel3_mean: 122.678914341
 * /movidius_ncs_nodelet/movidius_ncs_stream/cnn_type: alexnet
 * /movidius_ncs_nodelet/movidius_ncs_stream/device_index: 0
 * /movidius_ncs_nodelet/movidius_ncs_stream/graph_file_path: /opt/movidius/nca...
 * /movidius_ncs_nodelet/movidius_ncs_stream/log_level: 1
 * /movidius_ncs_nodelet/movidius_ncs_stream/network_dimension: 227
 * /movidius_ncs_nodelet/movidius_ncs_stream/scale: 1.0
 * /movidius_ncs_nodelet/movidius_ncs_stream/top_n: 3
 * /realsense_ros_camera/accel_fps: 1000
 * /realsense_ros_camera/accel_optical_frame_id: camera_accel_opti...
 * /realsense_ros_camera/color_fps: 30
 * /realsense_ros_camera/color_height: 480
 * /realsense_ros_camera/color_optical_frame_id: camera_rgb_optica...
 * /realsense_ros_camera/color_width: 640
 * /realsense_ros_camera/depth_fps: 30
 * /realsense_ros_camera/depth_height: 480
 * /realsense_ros_camera/depth_optical_frame_id: camera_depth_opti...
 * /realsense_ros_camera/depth_width: 640
 * /realsense_ros_camera/enable_color: True
 * /realsense_ros_camera/enable_depth: True
 * /realsense_ros_camera/enable_fisheye: True
 * /realsense_ros_camera/enable_imu: True
[... 71 lines cut ...]
[ INFO] [1521168446.172316837]: use channel3_mean = 122.679
[ INFO] [1521168446.173036455]: use top_n = 3
[ INFO] [1521168446.173884630]: use scale = 1
[ INFO] [1521168446.174020395]: open device #0 name=2.5
[ INFO] [1521168446.185153613]: setupDevice...
[ INFO] [1521168446.299520840]: Device Name: Intel RealSense 435
[ INFO] [1521168446.299595754]: Device Serial No: 727212071015
[ INFO] [1521168446.299631001]: Device Product ID: 0B07
[ INFO] [1521168446.300024468]: Device Sensors: 
[ INFO] [1521168446.300356490]: Stereo Module was found.
[ INFO] [1521168446.300421876]: RGB Camera was found.
[ INFO] [1521168446.300721174]: Fisheye sensor isn't supported by current device! -- Skipping...
[ INFO] [1521168446.300765074]: Gyro sensor isn't supported by current device! -- Skipping...
[ INFO] [1521168446.300794856]: Accel sensor isn't supported by current device! -- Skipping...
[ INFO] [1521168446.300879277]: setupPublishers...
[ INFO] [1521168446.485136723]: setupStreams...
[ INFO] [1521168446.487573425]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1521168446.487961655]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1521168446.488256032]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1521168446.491552604]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1521168446.508942561]: publishStaticTransforms...
[ INFO] [1521168446.509327956]: RealSense Node Is Up!
Booted 2.5 -> VSC

So please double check the environment on your side and find out which package is missing.

manhha1402, 2018-03-19:
Here is output of my computer:
+++++resource name: movidius_ncs_launch
-----resource location: /home/jcl-mh/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_launch
+++++resource name: rosgraph
-----resource location: /opt/ros/kinetic/share/rosgraph
... logging to /home/jcl-mh/.ros/log/c3c10f5c-2b4f-11e8-b25f-901b0e0c067c/roslaunch-jcl-stud-7478.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

+++++resource name: movidius_ncs_launch
-----resource location: /home/jcl-mh/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_launch
+++++resource name: realsense_ros_camera
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 748, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 720, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 654, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 684, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 588, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 204, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: realsense_ros_camera
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/jcl-mh/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share

I tried to use my own code with realsense + movidius stick, it worked. Here is the result of Alexnet:
[image]
I will check the missing packages.
 Another question: Can we draw the bounding box of detected object when using C/C++ to code for SSD mobile net ?

xhuan28, 2018-03-26:
Thanks for reminding. We don't always follow the latest commit of realsense ros package. We use tag 2.0.2 instead currently and will sync with realsense code accordingly if needed in future.
Currently, we don't implement RVIZ display, but it is in our plan.
Have you fixed the issue that you cannot obtain the image detection result? If yes, please provide more information and we will help you look at this.

manhha1402, 2018-03-26:
The problem is fixed. Thank you !

closed 2018-03-26 by manhha1402

patrickpoirier51, 2018-11-14:
Experienced this problem too with realsense d435 , as @manhha1402  wote ;
_I just check it out that the realsense node has changed its name to "realsense2_camera", so I think the movidius_ncs_lauch file should be modified as well._

I made it -run-with-other-ros-supported-cameras-1

roslaunch realsense2_camera rs_camera.launch
roslaunch movidius_ncs_launch ncs_camera.launch cnn_type:=tinyyolo_v1 camera:=others input_topic:=/camera/color/image_raw