intel/ros_intel_movidius_ncs #71: ROS No output?

ytgcljj, 2018-03-26:
I want to use ROS to display and detect the image in the ROS and how to modify it

ytgcljj, 2018-03-26:
[image]

xhuan28, 2018-03-26:
Did you mean there is no output in Rviz? The feature is in our plan, but not implemented yet. Currently, you can view the result through opencv viewer.

ytgcljj, 2018-03-27:
Yes, I want to do this, thank you.

ytgcljj, 2018-04-03:
@xhuan28 do it tell me

xhuan28, 2018-04-03:
We are planning to add this feature in next release. We also welcome contributions from community. So feel free to submit pull request to implement this feature if you like.

chaoli2, 2018-07-05:
Close this thread since the development plan is changed due to the release of [NCSDK v2](https://github.com/movidius/ncsdk/tree/ncsdk2).

closed 2018-07-05 by chaoli2