intel/ros_intel_movidius_ncs #74: cannot compile NCS graph using MobileNet_SSD model

ChampionRoy, 2018-03-30:
logs show as below:
`[libprotobuf ERROR google/protobuf/text_format.cc: 274] Error parsing text-format
caffe.NetParameter: 1177:17: Message type "caffe.LayerParameter" has no field has no field named "permute_param".`
And I do a little google, then I find this link [https://ncsforum.movidius.com/discussion/329/message-type-caffe-layerparameter-has-no-field-named-permute-param](url) 
Said that " Support for these layers is something we are looking into for a future release."
So, has this problem been fixed by now?

xhuan28, 2018-03-30:
which version of NCSDK are you using? Please update NCSDK to the latest version(v1.12).

ChampionRoy, 2018-04-02:
@xhuan28  Thanks for your response. I found that caffe dependency is not right. And finally I use this repository [https://github.com/weiliu89/caffe/tree/ssd](url)