intel/ros_intel_movidius_ncs #82: Image to CompressedImage?

yghcats, 2018-04-08:
In the   file : stream_detection.cpp
sensor_msgs type Image to CompressedImage?

thanks!

xhuan28, 2018-04-12:
do you mean your image type is compressedimage and if the project supports compressedimage type? could you please elaborate your question?

yghcats, 2018-04-17:
@xhuan28 
I want to use： “sensor_msgs::CompressedImage”  type data publish node，Not “sensor_msgs::Image” type

thanks!

chaoli2, 2018-07-05:
@yghcats 
We use [object_msgs](https://github.com/intel/object_msgs) as the default message interface. This interface is shared between several projects like [ros_object_analytics, etc](https://github.com/intel/ros_object_analytics). So it is not feasible to change the message type in our project.

Hence it is feasible to do this only in your own repository and keep it not upstream. Fork [ros_intel_movidius_ncs](https://github.com/intel/ros_intel_movidius_ncs) and [object_msgs](https://github.com/intel/object_msgs) projects and then do whatever you like to meet your requirement:)

closed 2018-07-11 by yghcats