intel/ros_intel_movidius_ncs #84: Running multiple NCS sticks simultaneously

fiorano10, 2018-04-12:
If I connect 3 NCS together, does object analytics launch make use all of them?

chaoli2, 2018-04-12:
@fiorano10 
Exactly! These 3 NCS will be used all by default.
Object analytics [launch](https://github.com/intel/ros_object_analytics/blob/master/object_analytics_launch/launch/analytics_movidius_ncs.launch) includes movidius NCS stream [launch](https://github.com/intel/ros_intel_movidius_ncs/blob/devel/movidius_ncs_launch/launch/includes/ncs_stream_detection.launch), which specify the default max_device_number is 255. That means we can use at most 255 devices in parallel by default. Just modify this parameter if you'd like to manually control the maximum number of devices.

fiorano10, 2018-04-12:
Awesome, thanks!

closed 2018-04-12 by fiorano10