intel/ros_intel_movidius_ncs #88: ncs_stream_detection.launch not publishing messages to the topic

fiorano10, 2018-04-19:
I have 2 Moviius NCS running on 16.04 Xenial, with Realsense D435, when I launch the ncs_stream_detection.launch file, it doesn't publish any messages to the topic, unless I restart it a couple of times. I generally use this with the object_analytics package to get object tracking and localization. Since it doesn't publish detection, there is no tracking or localization.

chaoli2, 2018-04-20:
@fiorano10 
May I know which command you use to launch the detection? Is there any error log when the topic doesn't publish any message?

fiorano10, 2018-04-20:
I'm on Version 0.6.0, I use `ncs_stream_detection.launch`, and there is no error log. This generally happens when I launch it along with the object_analytics package launch file.

fiorano10, 2018-04-20:
Also, when I run with 1 Movidius the detection I get is ~10-12 FPS, and when I connect another Movidius to the passive USB hub, the detection frequency remains the same. At rare occasions, the frequency with 2 Movidius would be ~18-20 FPS.

chaoli2, 2018-04-23:
@fiorano10 
There are several issues you mentioned in this thread. Thanks very much for your great interests on this project.  Let's clarify them one by one:

**1. You launch the ncs_stream_detection.launch file, but it doesn't publish any messages to the topic, unless you restart it a couple of times.**
It seems that you are facing some launching problems when you launch the stream detection node. Currently we also found one bug when terminating stream detection nodes (not sure if it is exactly the one you met). This bug will affect the following launches since some hardware resources are not released as expected. We've submitted patch to "devel" branch to fix it and it will be merged into “master” branch with next release. Will you switch your code to "devel" branch for a try and confirm whether your "launching problem" still exists?

**2. The FPS comparison between 1 NCS and 2 NCS**
According to our validation, the power supply for USB hub is important, since each NCS stick will consume certain power during computation. We connected two NCS and one camera to one passive USB hub, the camera were not even be detected due to insufficient power supply. So we strongly recommend you to use multiple NCS feature with active USB hub.

**3.Other issues with object_analytics project** 
We'll co-work with object_analytics project owners to solve any problem that may be caused by ros_intel_movidius_ncs project. We are now working in progress to identify the root cause about the problem of no message published with object_analytics . Once we confirm that this is caused by ros_intel_movidius_ncs , we'll update this thread more details.

chaoli2, 2018-07-05:
Close this thread since no response. Feel free to re-open this issue when need further assist.

closed 2018-07-05 by chaoli2