intel/ros_intel_movidius_ncs #95: Possible to run 2 tinyyolo_v1 and mobilenetssd simultaneously?

fiorano10, 2018-05-01:
Is it possible to run any 2 cnns' simultaneously on the same rgb stream from the realsense? I'm trying to run tinyyolo on 2 movidius, and mobilenetssd on the other 2. I would need to publish 2 movidius_detected_objects topics, right?

chaoli2, 2018-05-02:
@fiorano10 
This feature is not feasible currently. 
Assuming that you are using the lasted code on master branch - you have to launch two NCS nodes: one for ssd and another for tinyyolo. However, the multi-thread implementation for [multiple device support feature](https://github.com/intel/ros_intel_movidius_ncs/blob/master/doc/multiple_device.md) will block each other since we use std::thread::join() to join threads, which will block the main thread. 

We are now optimizing and improving thread management. Once this optimization is done, your requirement would be met by customizing launch files. As you mentioned, 2 topics are needed for two NCS nodes then.

chaoli2, 2018-07-05:
Close this thread since no response. Feel free to re-open this issue when need further assist.

closed 2018-07-05 by chaoli2