intel/ros_intel_movidius_ncs #96: Installation of NC App Zoo and where is the graph data

ahasenfr, 2018-05-05:
Hi there, 
I have trouble using the NC App Zoo.
In your guide under
> 3 Environment Setup

you write install NCSDK and install NC App Zoo. For the NCSDK there is a installation guide on its github site, but on the github site of NC App Zoo there is nothing like this. So I guessed that I should clone the repository somewhere on my machine (personal workspace) and then just do the linking to /opt/movidius/ncappzoo. Is that correct?

Furthermore you say, that after the installation I should make sure that i can find graph data in .../ncappzoo/caffee or /tensorflow which I can not find.

Can someone explain to me what I did wrong or have overlooked?

xhuan28, 2018-05-06:
Yes, you should clone the repository and create the linking to /opt/movidius/ncappzoo.  The graph file doesn't exist in ncappzoo by default, so you have to "cd" to one of the model folder and execute "make" command to build it out.
Take GoogLeNet for instance, please refer here(https://github.com/intel/ros_intel_movidius_ncs/blob/master/doc/image_classification.md#googlenet) for the detailed steps.

ahasenfr, 2018-05-06:
Okay, I tried to run the command "make all" in the directories "apps" and "caffee" and I got the same error message 

> ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type

So I searched a bit and found some information, that it is the "fault" of ROS. In some way it interferes with its 2.7 installation of python, even though python 3.5 has been installed through the NCSDK.
What can/should I do here?

In the end I just want to use the "IntelROSProject", didn't know that it is this hard/difficult to get it running :/

xhuan28, 2018-05-07:
Could you please try to exclude /opt/ros/kinetic/lib/python2.7/ from PYTHONPATH? I once found the same issue and after excluding the ros python path, it works.

ahasenfr, 2018-05-08:
How do I do that?
I'm new to unix systems :/

xhuan28, 2018-05-09:
export PYTHONPATH=/opt/movidius/caffe/python:$PYTHONPATH

ahasenfr, 2018-05-09:
Okay, thanks :)
In the directory of NC App Zoo I executed "make" to build it out.
But as I wanted to use [ros_moving_object](https://github.com/intel/ros_moving_object) I got an error.
This time not for graph, that worked fine, but for category_file_path.
The Error looks like this:
`[ERROR] [1525866428.240755098]: category_file_path = /opt/movidius/ncappzoo/data/ilsvrc12/voc21.txt not exists`

I looked in the folder and there was no voc21.txt.
Did just the name change in the meantime or is it something else?

closed 2018-05-14 by xhuan28