intel/ros_intel_movidius_ncs #97: category_file_path = /opt/movidius/ncappzoo/data/ilsvrc12/voc21.txt not exists

ahasenfr, 2018-05-09:
I'm making a new issue so that it isn't too messy.
I wanted to start the [ros_moving_object-Node](https://github.com/intel/ros_moving_object) and there it is required to start also "analytics_movidius_ncs.launch" which fails because of the error in the title.

So the voc21.txt does not exist in the mentioned folder and I don't know how to substitute this.
I looked up where the data folder originated from ( https://raw.githubusercontent.com/BVLC/caffe/master/data/ilsvrc12/get_ilsvrc_aux.sh ) and the source of the zipped directory (  http://dl.caffe.berkeleyvision.org/ ) "caffe_ilsvrc12.tar.gz" hasn't been updated since 12-Jul-2014.
And no file in the directory itself has been updated since then...

What should I do?

chaoli2, 2018-05-10:
@ahasenfr 
"[voc21.txt](https://github.com/intel/ros_intel_movidius_ncs/blob/master/data/labels/voc21.txt)" is located in the "[data/lables](https://github.com/intel/ros_intel_movidius_ncs/tree/master/data/labels)" directory of this project. 
Just following the last line of [4 Building and Installation](https://github.com/intel/ros_intel_movidius_ncs/tree/master#4-building-and-installation) to run: 
```cp ~/catkin_ws/src/ros_intel_movidius_ncs/data/labels/* /opt/movidius/ncappzoo/data/ilsvrc12/```
This command will copy all label texts to the default location to use.

ahasenfr, 2018-05-10:
Oh god, so sorry.
Must have totally overseen that :/

closed 2018-05-14 by xhuan28