intel/ros_intel_movidius_ncs #102: When does the package support the NCS2.0?

MengNan-Li, 2018-08-13:


chaoli2, 2018-08-14:
@MengNan-Li 
We will support the latest NCSDK as a part of Intel [OpenVINO ](https://software.intel.com/zh-cn/openvino-toolkit)in future. So we do not have a plan to support NCSDKv2 yet.

MengNan-Li, 2018-08-14:
@chaoli2 
OK, Thinks. Does this‘s mean the intel OpenVINO will include NCSDK and ncappzoo? And, can we use the [ros_object_analytics](https://github.com/intel/ros_object_analytics) function in the OpenVINO?

chaoli2, 2018-08-14:
@MengNan-Li 
Intel OpenVINO will include new NCSDK. For ncappzoo, we didn't receive certain information that it will be included or not, but I guess it will :)

When the ROS/ROS2 OpenVINO project is done, we'll add its support for [ros_object_analytics](https://github.com/intel/ros_object_analytics) and [ros2_object_analytics](https://github.com/intel/ros2_object_analytics). So you can use their function in OpenVINO in the future. We may first support ROS2 version earlier than ROS.

MengNan-Li, 2018-08-14:
I got it. Thinks again!

closed 2018-08-14 by MengNan-Li