intel/ros_intel_movidius_ncs #107: could not convert ‘{{<expression error>, "device is busy, retry later"},

bruceEeZhao, 2019-04-22:
Following these instructions:
cd ~/catkin_ws/src
git clone https://github.com/intel/object_msgs
git clone https://github.com/intel/ros_intel_movidius_ncs.git
cd ros_intel_movidius_ncs
git checkout master
cd ~/catkin_ws
catkin_make

I met errors:
```
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:29:5: error: ‘MVNC_BUSY’ was not declared in this scope
     MVNC_BUSY,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:34:5: error: ‘MVNC_ERROR’ was not declared in this scope
     MVNC_ERROR,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:39:5: error: ‘MVNC_OUT_OF_MEMORY’ was not declared in this scope
     MVNC_OUT_OF_MEMORY,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:44:5: error: ‘MVNC_DEVICE_NOT_FOUND’ was not declared in this scope
     MVNC_DEVICE_NOT_FOUND,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:49:5: error: ‘MVNC_INVALID_PARAMETERS’ was not declared in this scope
     MVNC_INVALID_PARAMETERS,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:54:5: error: ‘MVNC_TIMEOUT’ was not declared in this scope
     MVNC_TIMEOUT,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:59:5: error: ‘MVNC_MVCMD_NOT_FOUND’ was not declared in this scope
     MVNC_MVCMD_NOT_FOUND,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:64:5: error: ‘MVNC_NO_DATA’ was not declared in this scope
     MVNC_NO_DATA,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:69:5: error: ‘MVNC_GONE’ was not declared in this scope
     MVNC_GONE,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:74:5: error: ‘MVNC_UNSUPPORTED_GRAPH_FILE’ was not declared in this scope
     MVNC_UNSUPPORTED_GRAPH_FILE,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:79:5: error: ‘MVNC_MYRIAD_ERROR’ was not declared in this scope
     MVNC_MYRIAD_ERROR,
     ^
/home/bruce/catkin_ws/src/ros_intel_movidius_ncs/movidius_ncs_lib/src/exception_util.cpp:82:1: error: could not convert ‘{{<expression error>, "device is busy, retry later"}, {<expression error>, "an unexpected error was encontered during the function call"}, {<expression error>, "the host is out of memory"}, {<expression error>, "there is no device at the given index or name"}, {<expression error>, "at least one of the given parameters is invalid in the context of the function call"}, {<expression error>, "timeout in the communication with the device"}, {<expression error>, "the file named MvNCAPI.mvcmd should be installed in the mvnc direcotry"}, {<expression error>, "no data to return"}, {<expression error>, "the graph or device has been closed during the operation"}, {<expression error>, "the graph file may have been created with an incompatible prior version of the Toolkit"}, {<expression error>, "an error has been reported by Movidius VPU"}}’ from ‘<brace-enclosed initializer list>’ to ‘std::map<int, std::__cxx11::basic_string<char> >’
 };

```

chaoli2, 2019-04-22:
@bruceEeZhao 
It seems that you are using wrong NCSDK version.
Will you please re-install your NCSDK with below tag and try again?
```
git clone https://github.com/movidius/ncsdk
cd ncsdk
git checkout v1.12.00.01
make install
```

bruceEeZhao, 2019-04-24:
yes,taht works!!

closed 2019-04-24 by bruceEeZhao