intel/ros_intel_movidius_ncs #108: detection error

yghcats, 2019-05-17:
I run: 
roslaunch movidius_ncs_launch ncs_camera.launch cnn_type:=tinyyolo_v1 camera:=usb
roslaunch movidius_ncs_launch ncs_stream_detection_example.launch camera_topic:="/usb_cam/image_raw"

Error messages will appear after a period of time:

terminate called after throwing an instance of 'movidius_ncs_lib::MvncError'
  what():  an unexpected error was encontered during the function call
[movidius_ncs_nodelet/movidius_manager-2] process has died [pid 6435, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=movidius_manager

chaoli2, 2019-05-17:
@yghcats 
May I know your [NCSDK ](https://github.com/movidius/ncsdk)version? Please use tag [v1.12.01.01](https://github.com/movidius/ncsdk/tree/v1.12.01.01) instead of master or v2.xx

yghcats, 2019-05-17:
@chaoli2 My NCSDK is version v1.12.00.01 (NCSDK-1.12.00.01.tar.gz)