intel/ros_intel_movidius_ncs #109: the future plan of the NCSDK

hz658832, 2019-06-12:
Hey guys,
I find your work of the inferece within the ros project most interesting so I tried to convert this package from NCSDKv1 to v2 and it works.
With 2 Movidius Sticks I can achive detection 18Hz with MobileNetSSD, tracking about 10Hz and lokalization 18 Hz. My project uses intel NUC as host and ZED stereo as cam, the point cloud was filtered sparse for more performance.
Do you guys have future plan for this project? Will you continue updating the package, or you just put more focus on openvino inference, which is more flexible and could be more efficient.

Thanks,

Bests,

Haoming

chaoli2, 2019-06-13:
@hz658832 
Glad to hear that you are interested in this project. 
As you know, we indeed put most of our resources on OpenVINO ROS package. This project will be maintained in low priority, mainly for bug fixing.

According to our test, the early version of OpenVINO (2018_R3 and 2018_R4) support both NCS hardware v1 (MYRIAD 2) and NCS2 hardware v2 (MYRIAD X). But from 2018_R5, OpenVINO only support NCS hardware v2 (MYRIAD X). Just for your reference if you'd like to try it.

hz658832, 2019-06-13:
> @hz658832
> Glad to hear that you are interested in this project.
> As you know, we indeed put most of our resources on OpenVINO ROS package. This project will be maintained in low priority, mainly for [...]

Hey Chao,
thanks for your reply. I just went through your code for ros openvino, and I like your structure of the code. Actually i am starting with implementing the yolov3 inheriting from your yolov2 inference on this project. If you are interested so we could work together on this project. 

For OpenVino, I always trend to use the newest version of openvino, after suffering from the new structure of the sdk, i also got the code running on OpenVINO 2019_R1.1.

Kinds,

Haoming

chaoli2, 2019-06-13:
@hz658832 
Sure. Your contribution to OpenVINO package is highly welcome~ 
Let's keep in touch for any issue when you are implementing yolov3. Hope to see your PR in the future :)